Snapmaker/Luban #2471: Bug: Luban prevents the display from turning off

connor4312, 2024-03-31:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.11.0

### Is this a regression? (optional)

Unknown

### To Reproduce
Steps to reproduce the behavior:
1. Be on Windows
2. Open Luban. It doesn't matter whether it's in the foreground or minimized
3. In an administrative powershell, run `powercfg /requests`
4. See that Luban is keeping the display on 🐛 

```
PS C:\WINDOWS\system32> powercfg /requests
DISPLAY:
[PROCESS] \Device\HarddiskVolume3\Program Files\Snapmaker Luban\Snapmaker Luban.exe
Electron
```

Expected behavior

Luban should not keep the screen on, especially when it's not in the foreground

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 11
- Printer: N/A

</code></pre>

darienm, 2024-05-16:
FYI: This issue also affects Luban version 4.12.0 running on MacOS.
[image]

suspicious-pineapple, 2025-02-18:
i just commented out the line in src/main.js, that fixed it for me

[image]

darienm, 2025-02-18:
> i just commented out the line in src/main.js, that fixed it for me
> 
> [image]

Confirmed, this alters the flag on macOS as well (different line number). Thanks!