Snapmaker/Luban #2480: Bug: Not starting on Ubuntu Ubuntu 24.04 LTS

Flattermann78, 2024-07-26:
# 🐞 bug report

### Affected Version(s)
Luban 4.13.0 ist not starting on my Ubuntu 24.04 LTS. Command line reports

LaunchProcess: failed to execvp:
/opt/Snapmaker

and it claims that a memory dump is written.

### Is this a regression? (optional)

Not that I'm aware of.

 Yes, the previous version in which this bug was not present was: ....

First encounter

### To Reproduce

I only installed the new version and it failed. Unfortunately, I haven't tried the old version after system update.

 A clear and concise description of the problem...

Expected behavior

The Luban should start :-)

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Ubuntu 24.04
- Printer:  Snapmaker A250

</code></pre>

aiurlano, 2024-07-31:
It's happening for me as well after upgrading to Ubuntu 24.04 LTS.
I had Luban 4.12.0 installed which was working fine before upgrade.
After ubuntu upgrade, I can confirm both 4.12.0 and 4.13.0 are not starting with the error:
> :~$ snapmaker-luban 
> LaunchProcess: failed to execvp:
> /opt/Snapmaker
> Trace/breakpoint trap
> [38282:0100/000000.684022:ERROR:zygote_linux.cc(646)] write: Broken pipe (32)
>

aiurlano, 2024-07-31:
After some investigation I found out that it is most likely caused by this [issue](https://github.com/electron/electron/issues/41066) which is hitting several electron apps.
It seems related to apparmor Unprivileged user namespace restrictions that are now enabled by default in Ubuntu 24.04

Disabling the sandbox make Luban start:
`snapmaker-luban  --no-sandbox --disable-gpu-sandbox`