Snapmaker/Luban #2490: Bug: Cannot start jobs with absolute positioning when using Snapmaker Ray and when not connected to the machine in the beginning.

McToel, 2024-09-21:
# 🐞 bug report

The generated gcode contains motor_mode: 1 which resets the cords and then jobs cannot start. Boundary runs fine.

### Affected Version(s)
latest

### To Reproduce

1. Start Luban
2. Do not connect to machine
3. Add something to run for example the following: 
[image]

4. Generate gcode, load gcode to workspace and connect to Ray
5. Use Control Mode
6. Try to start job, which will result in the corsair to blink and nothing more. The job will not start.

### Debug info
Logs show the following error
```
[I][MPSVC]start SD file: /ray.nc
[I][SNMKR]SYS STA pre: STARTING, req: PRINTING, now: PRINTING
[I][FMGR]Plat SD started!
[I][SNMKR]got header start
<Idle|WPos:0.000,0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000,0.000|SD:0.61,ray.nc>

[I][SNMKR]total file line: 134
[I][SNMKR]estimated time: 19s
[I][SNMKR]motor mode: 1
[I][MPSVC]last motor mode = 3, new motor mode = 1
[I][PLAT]reset sys pos and WCO because disabling motor
[I][PLAT]Reset during SD file at line: 26
[I][SNMKR]plat alarm: 3
[E][FMGR]Plat alarm in working, stopped!
[I][SNMKR]SYS STA pre: PRINTING, req: COMPLETED, now: COMPLETED
ALARM:3
[I][SNMKR]SYS STA pre: COMPLETED, req: IDLE, now: IDLE
[I][FMGR]set file mgr status, now: 7, new: 0
[I][FMGR]print done!
```

Gcode contains motor_mode: 1 in line 26

### Solution

I found that there is a check to set motor mode 1 if machine is not connected.

https://github.com/Snapmaker/Luban/blob/ab5e8fb6bb5aa9b270a63718e1164549c8332111/src/app/flux/editor/actions-process.ts#L437-L441

I suggest removing that line, as it fixes the problem and I don't see any usefulness of the check. Please tell me if I'am wrong or if I should make a PR

McToel, 2024-09-21:
Some further investigation tells me that in might also be the `SetupCoordinateMethod` is `Manual` by default. I guess that is kind of correct, however the side effect which I am experiencing is not. In order to reliably start jobs, it seems that I need to open workspace first, then set the control mode, and then I can open my laser project and slice it.

McToel, 2024-09-21:
Maybe one solution might be to add the gcode headers not while generating gcode but when exporting the gcode either to file or to the machine (not when exporting to workspace thought).

xiwai2, 2024-09-26:
Hello, in the latest version of Luban and Ray, A new control mode is added in order to use the A-B Position function, but due to some limitations, the Gcode of control mode and manual mode are distinguished. When Luban generates Gcode in manual mode or when the machine is not connected, it is limited to manual mode jobs, and when the connected machine switches mode to control mode, it generates Gcode again only to control mode jobs.

McToel, 2024-09-26:
In my case, Luban does not regenerate the gcode if I set control mode. Or at least it still exports gcode with the wrong motor mode to the machine. Maybe the regeneration is not working and causing the problem.

Anyway, I find that it's a common use case to work on workpieces that have the exact dimensions of the working area, and therefore control mode is definitely the best option, and it's currently unusable. I needed more than two days to troubleshoot the issue and only found it because I can read the logs and then dug around in the sources code, which cannot be expected from all users of Ray laser cutters.

xiwai2, 2024-09-26:
Sorry, maybe I didn't express it properly. If you want the machine to work in control mode, you can connect the machine to Luban, switch to control mode on Luban, and then slice.

closed 2024-09-27 by xiwai2