Snapmaker/Luban #2494: Bug: Excessive jogging in CNC with disjoint paths

dgatwood, 2024-09-29:
# 🐞 bug report

### Affected Version(s)

All

### Is this a regression? (optional)

No

### To Reproduce

1.  Create a CNC program that carves several paths that don't overlap, e.g. etching text into something.
2. Run the program.

Expected behavior

I expect Luban to realize that the paths are entirely disjoint, and generate Gcode that etches one letter followed by the next to minimize the amount of unnecessary jogging.  Instead, Luban generates Gcode that etches everything all at once.

This is unfortunate, for several reasons:

- The more jogging you do, the more error will creep into the position of the hardware, and the more likely you are to have small bits bind in the material and snap off.
- The more jogging you do, the more time you waste.

The two-cut program that I'm running right now was going to take two hours and 16 minutes, if memory serves.  I split it into two separate programs, one per cut, and it is now going to take an hour and forty-two minutes.

For a three-axis model, there is absolutely no reason to carve every independent path at a given layer simultaneously unless you're doing something terrifying like setting the Z origin below the top of the original physical material.  After all, the CNC cutter can always just go higher than the physical material, jog, and start over.

It could theoretically save time to do so if and only if the paths are non-disjoint, because you could avoid even cutting parts of a path where the material is already completely gone, but Luban also does not do that optimization as far as I can tell, making this point moot.

In an ideal world, Luban would identify disjoint paths and cut them independently, then identify overlapping paths and cut them together, avoiding re-cutting areas that were already carved out by another path.  That said, I'd settle for Luban just identifying disjoint paths and cutting them independently so that it doesn't waste quite as much time jogging back and forth above the material when it could be cutting. 

## 🌍  Your Environment

**Platform:**
- Operating System: macOS Sonoma
- Printer: A350

dgatwood, 2024-10-09:
Speeding up the jogging kind of misses the point, though, which is that there is no reason to do the jogging in the first place.  And even at the maximum speed and minimum height (which I was using when I got those numbers), it still adds up to a lot of wasted time.

xiwai2, 2024-10-12:
Hello, could you provide us with a Gcode that you think is too many times for us to check the problem? Thank you

xiwai2, 2024-10-12:
Hello, can you provide us with a Gcode that you think has too many jogs to check the problem? Thank you

alexastro41, 2025-01-02:
I have run into a similar problem. Using Luban, I wanted to engrave text into a 1" x 4". Using on the path or outline modes would have taken a total time of 7 - 10 minutes. When I used fill it took 3-4 hours. Now I would have used on the path or outline but there would have been a very small sliver of material inside of the letters. How I would have assume this would have worked is the gcode would have done the outline of each of the letters and then moved in or out and did a parallel path. Given the time difference this would have resulted in about 30 minutes at most. Instead it milled like a printer sweeping left and right plunging to make tiny dots to "fill" the letters. Maybe there needs to be more options inside of fill or the other modes? But until such a feature exists I will be using a different program to engrave letters with CNC milling.