Snapmaker/Luban #2551: Bug: xxxx in 3d printing

phil3412, 2024-12-25:
# 🐞 bug report

### Affected Version  4.14.0

 The issue is caused by version 4.14.0  latest

### To Reproduce
Steps to reproduce the behavior:
1- load a model (vase)
2-select vase mode in model structure
[image]
[image]
[image]
3-generate gcode
 look at the extruder that goes thru the printed vase (see pictures)
and the printed vase is same as the preview... 
[image]

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 11 x64,
- Printer: j1s

</code></pre>

phil3412, 2024-12-25:
all printed with left extruder, that will extrude from left to right but not from right to left  when the extruder pass thru le model

xiwai2, 2025-02-27:
[image]

[image]
Hello, I tried to reproduce your problem with any model, but it didn't work. Can you provide me with your model file?