Snapmaker/Luban #2581: LunarTPP source code

aiurlano, 2025-01-23:
Hello. I'm having a lot of problems with SVG files in Luban when doing laser engraving projects.
Luban often fills the paths and shapes even when fill:none is specified in the SVG file. I was not able to find a workaround for this yet, so I'm unable to laser engrave such files. All the tests I made resulted in wasted material.

After investigating the source code for a couple of days, I found out that the SVG is preprocessed by Luban and then passed to the LunarTPP program, which returns what seems to be a list of toolpaths that is later converted to a gcode file.
Currently this LunarTPP program seems to completely ignore the stroke-width and fill properties of svg objects.

I wanted to have a look in case it was something trivial to fix, so I downloaded the snapmaker-lunar repository, but I found out that it does not contain the LunarTPP source code.

It does not seem to be closed source software, as the LICENSE is AGPL.

Could someone please point me to the source code of LunarTPP?

Thanks

xiwai2, 2025-02-26:
您好，看到您已经从我们售后服务那里收到回复了，这面先给你关闭问题

closed 2025-02-26 by xiwai2

aiurlano, 2026-01-16:
Snapmaker never provided me with the source or indicated where I could download the source code from.

I found out this repository [https://www.npmjs.com/package/@snapmaker/snapmaker-lunar](https://www.npmjs.com/package/@snapmaker/snapmaker-lunar) that contains the LunarTPP binary file but not the source code. I thought that the AGPL license meant that all the source code must be included, but maybe I am mistaken?

It looks like development on Luban stopped one year ago and laser engraving and cutting is still mostly unusable with complex vector files.

I made my own replacement program for my very specific use case that worked with what I needed to do at that time only but I'm not willing to release its source to help a company that behaves like this